ryaniftron/clearview_interface_public #22: ESP32 reboots after bad command

HazardCreative, 2020-05-09:
To reproduce:

1. Boot ESP32 and connect to MQTT server
2. Start a RotorHazard server

```

ryaniftron, 2020-05-10:
From initial tests, this seems to be caused by something on the ESP32's RX buffer. Without hooking up a ClearView, the issue is not present. Due to an exception error not letting all the debug messages through, although it looks like failing during the UART TX, it could be on the return parsing.

ryaniftron, 2020-05-10:
When testing with a ClearView, I was not able to reproduce the error. The only command sent is ODD to turn off the OSD when the server starts up. This does not ask for a reply from the CV, so there is nothing in the RX buffer to get handled improperly. Perhaps you aren't running the latest EPS32 code?

HazardCreative, 2020-05-11:
I'll update. But does the ESP reboot after any bad command? It should not ignore malformed commands, not fail and reboot.

ryaniftron, 2020-05-11:
No, the ESP takes the payload from a command and sends it directly to the CV. If the CV responds to a report request, the ESP packages up the response (whatever it is) and ships it back to RH.

All the checking for malformatted commands and responses occurs in the python code. Thus, I am not clear what would cause a reboot other than some overflow or memory access issue. I'll keep an eye on it though.

ryaniftron, 2020-05-13:
Closing because there does not appear to be this bug anymore.

closed 2020-05-13 by ryaniftron