ryaniftron/clearview_interface_public #57: ESP32 - Make UI look nice

ryaniftron, 2020-09-01:
- Add some classes for display
- Make it a bit more responsive on re-org
- Padding for buttons
- Maybe a two column layout

HazardCreative, 2020-09-06:
Contributing the following to be considered for the CVCM UI.
[cvcm-interface.zip](https://github.com/ryaniftron/clearview_interface_public/files/5179038/cvcm-interface.zip)

Includes:
- HTML improvements
    - remove style tags
    - improve structure
    - better use of HTML elements
- CV website inspired styling
- JS improvements
    - remove use of direct styles
    - disable fields while submitting
- supporting SVG assets

ryaniftron, 2020-09-10:
@HazardCreative Thank you very much for the contribution. It looks awesome.

1. For some reason, the <main> tag is getting a success styling applied at some point. You can see the little green check up in the corner. Any ideas what's causing it? 
[image]

1. I'd like to use the style of "Current Status" that appears in `subtitle.html`  in the test tab for advanced options. I changed from referencing the id of "current_status" to creating a "status" class in the css file, but it does not appear to apply to the advanced settings (see `test.html`). 

Once these issues are fixed, this code can be merged and I'd like to publish a new release from it.

ryaniftron, 2020-09-17:
Remaining work: minify and add in-progress loading asset.